Elecrow-RD/CrowPi2 #3: cant boot

fase1, 2020-11-09:
just received the crowpi2, made a backup from enclosed sd-card, installed pasi 4B 8GB and tried to boot.
unfortunatly the crowpi wont boot from original sd card nor from my backup with error "FAT volume not supported"
- bootloader 940f978d Mar 19 2020

any ideas or maybe a source to get a working sd card image?

Pearl-852, 2020-11-09:
Perhaps download the offical Raspberry Pi OS image, burn it to the microSD card and then install the custom software following the instructions at the Wiki section [here](https://github.com/Elecrow-RD/CrowPi2#to-install-the-new-raspbain-system-what-configuration-should-be-done)

fase1, 2020-11-09:
thx for this, i did it. but where can i find the learning software start page? just stuck in the raspiberry os desktop...

Elecrow-RD, 2020-11-10:
> thx for this, i did it. but where can i find the learning software start page? just stuck in the raspiberry os desktop...

1. Before backing up the image, you need to format your MicroSD card with the FAT32 format.
2. The method that installing the new Raspbian image means boot up to standard Raspberry Pi OS desktop but not boot up to CrowPi2 OS, there are different, you can't boot to CrowPi2 OS with configuration. And that is why we ask users to back up the CrowPi2 OS after receiving the CrowPi2.

By the way, any news about releasing image, we will update to our official website(https://www.elecrow.com/)

closed 2020-11-10 by Elecrow-RD

fase1, 2020-11-10:
But you just can make a usefull backup if the source is working!

Pearl-852, 2020-11-10:
@fase1 This [post](https://www.kickstarter.com/projects/elecrow/crowpi2-steam-education-platformand-raspberry-pi-laptop/comments?comment=Q29tbWVudC0zMDU0NDI3NQ%3D%3D&reply=Q29tbWVudC0zMDU0NTU0NA%3D%3D) in Kickstarter from Jon Wager should help with your situation.