Elecrow-RD/CrowPi2 #4: Cannot input the >, <, | characters on the German keyboard

Pearl-852, 2020-11-10:
My keyboard layout settings are
[image]

But the key labeled for >, <, | characters does not seem to function, all other german special characters (i.e. umlauts or ß) works. When the layout is set to US keyboard, the same key works as 'CTRL', so the key is ok. 

So, is my keyboard settings shown above correct? If not, please advise the proper settings.

Pearl-852, 2020-12-03:
This issue has been opened since Nov.10 and there hasn't been any official comment on here. If you check the Comments section on Kickstarter, there are numerous reports with this same issue since.

Please acknowledge.

Jay-Kopf, 2020-12-15:
Please fix this. It's a development device and with these characters it is problematic to write code or work with linux bash!

Pearl-852, 2020-12-16:
FYI, as a _temporary_ measure for those **3** missing characters, I enter them as Unicode character number as follows

> **<** : Press **Shift**+**Ctrl**+**U**, release, then type **3c** (ascii 60) and press **Enter**
> **>** : Press **Shift**+**Ctrl**+**U**, release, then type **3e** (ascii 62) and press **Enter**
> **|** : Press **Shift**+**Ctrl**+**U**, release, then type **7c** (ascii 124) and press **Enter**

And, I am also investigating/studying a way to use **_xmodmap_** to remap that Control_R (labeled on the US keyboard) key.

X-Perry-Mental, 2021-02-19:
I don't know if it is still relevant to you, but for me, the German keyboard is working flawlessly. My CrowPi2 just arrived today, and all I did was setting up the device using the terminal:

sudo raspi-config

Within the localisation options (5) I have set:
L1 to "de_DE.UTF-8 UTF-8"
L2 to "Europe" / "Berlin"
L3 to "Generic 105-key PC (intl.)" / "German" / AltGr to default / no compose key / Ctrl+Alt+Backspace: no

I can see from your screenshot that you did use the graphic tool and not the terminal, maybe that’s making a difference.
Apart from the settings above I updated the system using apt, but that’s all I did so far.

Best regards, Perry

Pearl-852, 2021-02-21:
@X-Perry-Mental 
Thanks for your feedback and glad to hear your German keyboard is working flawlessly. I have just edited my keyboard settings following your instructions using ` sudo raspi-config `, and confirmed as shown below

`$ cat /etc/default/locale`
```
#  File generated by update-locale
LANG=de_DE.UTF-8
LC_ALL=de_DE.UTF-8
LANGUAGE=de_DE.UTF-8
```
`$ cat /etc/default/keyboard`
```
# KEYBOARD CONFIGURATION FILE

# Consult the keyboard(5) manual page.

XKBMODEL="pc105"
XKBLAYOUT="de"
XKBVARIANT=""
XKBOPTIONS=""
```
Unfortunately, my result remain the same, not able to type those 3 missing characters. Since you have just received your CrowPi2, I have a hunch that Elecrow may have revised the keyboard firmware to fix this issue. Because I believe this is/was a keyboard firmware issue.

I would appreciate if you can do the follow test
1. Run ` cat /etc/default/keyboard ` and post your result.
2. Run ` sudo showkey -a ` and test those 3 characters **< > |** and post the result.

Thank you very much.

X-Perry-Mental, 2021-02-22:
pi@raspberrypi:~ $ cat /etc/default/keyboard
# KEYBOARD CONFIGURATION FILE

# Consult the keyboard(5) manual page.

XKBMODEL="pc105"
XKBLAYOUT="de"
XKBVARIANT=""
XKBOPTIONS=""

BACKSPACE="guess"

---

pi@raspberrypi:~ $ sudo showkey -a
< 	 60 0074 0x3c
> 	 62 0076 0x3e
| 	124 0174 0x7c

Best,
Perry

Pearl-852, 2021-02-22:
@X-Perry-Mental 
Thank you so much for testing.

If you don't mind, I would like to confirm whether or not your keymap (de) file has been customized by Elecrow. Please follow the steps below to attach that keymap file to your new reply post
1. Open a new Terminal.
2. Type `cp /usr/share/X11/xkb/symbols/de ~/de.txt`
3. Open your File Manager, then drag & drop the copied file `de.txt` onto a new post here.
4. You may delete that copied file, `de.txt`

Thanks,

X-Perry-Mental, 2021-02-22:
Here we go, hope it helps!
Best,
Perry
[de.txt](https://github.com/Elecrow-RD/CrowPi2/files/6021612/de.txt)

Pearl-852, 2021-02-22:
@X-Perry-Mental 
Thanks for posting the file. The file contents are identical, so there is no customization there.

**Conclusion:**
IMHO. The German keyboard controller firmware must have been updated to correct this issue. Now is upto Elecrow on how they will help their affected beckers to resolve this issue.

atantnita, 2021-02-23:
Hey,
I think, there is a firmware-error, see below for "|". My workarounds are:
In X11 open terminal and enter
> xmodmap -e "keycode 105 = less greater"
and then < (normal) and > (with shift) work. Unfortunately this doesn't work with |, and so I  substitute for the EURO-symbol alt key e
> xmodmap -e "keycode 26 = e E e E bar"
You can put both commands into Your .bashrc, and then the keys (< as key, > as shift <, and | as altgr-e) are available.

If You want to use these keys in console-mode (alt-F1 .. alt-F6) then use
> loadkeys myckeys
where myckeys is a file containing following three lines (with space before 2nd and 3rd)
keycode 97 = 0x003c
 shift keycode 97 = 0x003e
 altgr keycode 97 = 0x007c

Then You can use the three symbols in console mode

Best regards

manfred

Pearl-852, 2021-02-24:
Manfred, 
Great job! Thanks for your sharing.

davidrichardzh, 2022-10-23:
SOLVED: ACTUALLY THERE IS A KEYBOARD (GERMAN NO DEAD KEYS), THE PROCEDURE IS:

OPEN Terminal and type: 
$ sudo raspi-config [enter]
Then pick the following
5. Localisation Option [enter]
Keyboard [enter]
Generic 105-keyPC (intl.) [enter]
German – German (no dead keys) [enter]
The default for the keyboard layout [enter]
No compose key [enter]
<YES>
<Finish>

CHECK THAT YOU SEE 1. THE FOLLOWING
$ cat /etc/default/locale [enter]
#LANG=C.UTF-8
#LC_ALL=zh_CN.UTF-8
#LANGUAGE=en_US:en
#LANGUAGE=en_US:en

and CHECK THAT YOU SEE 2. THE FOLLOWING too:
$ cat /etc/default/keyboard [enter]
XKBMODEL="pc105"
XKBLAYOUT="de"
XKBVARIANT="nodeadkeys"
XKBOPTIONS="terminate:ctrl_alt_bksp"
BACKSPACE="guess"

Works on my CrowPI2 purchased on Oct 20, 2022