Elecrow-RD/CrowPi2 #5: Schematics

rabagliati, 2020-11-13:
Please can we download the schematics ?

Much of the online documentation refers to the CrowPi board, which I think had DIP switches to turn on sensors.

What does the Switch to the left of the fan do ? I think it is a double pole switch for I2C bus ?

Power supply - I have done this :-

[image]
[image]
[image]

Is this github page the user forum?

I have also found a user forum, not very active, around CrowPi.

http://www.crowpi.co.uk/phpBB3/index.php

Elecrow-RD, 2020-11-20:
> First, I would like to say that the Crowpi2 is very nice - improvements over the Crowpi1 include being able to put jumper wires to the RPi connector while it is plugged in, and clear markings of sensor addresses and pin numbers (Wpi numbering on the connector) make it easy to use.
> 
> There is source code for all the examples, so you find the one that uses your sensor/driver and look at the code. Thank you to Electrow for the attention to detail.
> 
> Please can we download the schematics ?
> 
> Much of the online documentation refers to the CrowPi board, which I think had DIP switches to turn on sensors.
> 
> What does the Switch to the left of the fan do ? I think it is a double pole switch for I2C bus ?
> 
> Power supply -
> 
> Looking at the chips used on the board powering the Rpi it might easily handle 18V input.
> 
> But there are other power components around the board, so I used a linear regulator (which gets hot - hence large heatsink).
> 
> I have done this :-
> 
> [image]
> 
> [image]
> 
> Is this github page the user forum?
> 
> I have also found a user forum, not very active, around CrowPi.
> 
> http://www.crowpi.co.uk/phpBB3/index.php
> 
> Schematics, please ?

The switch is on the left of the fan is for switching the connection between on-board sensors and 40-pins header. That is, toggle the switch to the left to make 40 pins of raspberry pi connect to on-board sensors; while toggling to the right to disconnect the on-board sensors and then you can use pins from the 40-pins header to connect extra sensors or module that are not on CrowPi2, or doing the experiment in the breadboard.

We have not yet opened a forum, please be careful to prevent fraud

Sorry, the circuit diagram is not convenient to provide, which involves intellectual property protection!

closed 2020-11-20 by Elecrow-RD

Pearl-852, 2020-11-20:
Understood your IP rights and we respect that, but is it possible just open the section for the **Switch** circuit schematic ONLY. It would be a tremendous help for our understanding to prevent malfunctions when wiring our own project on the breadboard.

After all, your **CrowPi2** is being market as a learning platform.

SpareSimian, 2021-11-22:
What IP is involved? Open-source hardware has the same advantages for users as open-source software: It allows us to fix problems as a community. I just got a CrowPi2 as a birthday present, my first Pi, and I'm fighting a few issues that might be hardware or software. Having schematics would allow to rapidly identify which is which. For example, I'm having trouble keeping the wifi connected, which might be an RF problem. I'm getting no sound, which might be a driver issue, or it might be a loose connection somewhere. (I build and program industrial robots for a living so I've had decades to fight such problems. Most of my issues end up being bad connections.)

SpareSimian, 2021-11-22:
BTW, the CrowPi2 reminds me of the Heathkit microprocessor trainer from around 1980. It has the same feel of hardware and software. But the Heathkit included full schematics. One of my college courses used those schematics to teach us how computers worked from an electronics perspective. It was probably the most valuable course of my career. So your schematics could make great courseware.

texsean, 2026-11-14:
I have one that the power supply of smoked with wrong cable to the usbc.. I got replacement.. but hate that just the softboot chip bricked a otherwise good pi 5 learning tool and display. Does anyone have a schematic that i could hardwire a on off power suppluly cable to it?

[image]

texsean, 2026-11-14:
Softboot chip lol.