Elecrow-RD/CrowPi2 #7: The Function (Fn) key on US keyboard has no function at all

Pearl-852, 2020-11-16:
On the original Raspberry Pi Keyboard and the new Raspberry Pi 400, the **Fn** key + a key combo has the following KeyMap,
* Page Up: Fn+↑ (UpArrow)
* Page Down: Fn+↓ (DownArrow)
* End: Fn+→ (RightArrow)
* Home: Fn+← (LeftArrow)
* Insert: Fn+Del
* 0: Fn+M
* 1: Fn+J
* 2: Fn+K
* 3: Fn+L
* 4: Fn+U
* 5: Fn+I
* 6: Fn+O
* 7: Fn+7
* 8: Fn+8
* 9: Fn+9
* +: Fn+;
* -: Fn+P
* *: Fn+0
* /: Fn+/

I would recommend to provide your own custom keyboard driver with the above additional KeyMap and revise the key-cap silkscreening.

Pearl-852, 2020-11-16:
Just received a reply on the Kickstarter - CrowPi2 project page that only **Fn**+**F1** is assigned for toggling the Touchpad. But I still recommend to add at lease the first 5 key combo as above if possible. Those 5 key combo assignments are extremely helpful when writing codes and working in the Terminal (cli).

reostat, 2020-11-16:
I concur; code editing without Home and End keys is terribly awkward; other bindings are quite useful, too. Custom keyboard layout or key mapping would be awesome.

reostat, 2020-11-16:
I did some poking around and now I'm not sure that even a custom driver will cut it.

Thing is, Fn key is not a "normal" key, there's no scancode for it; instead, having it pressed down makes keyboard to send a different scancode for the key it's been pressed with. So I used `showkey` in terminal to verify if I would get different scancodes for keys pressed with or without Fn; alas, I get the same scancodes.

Okay, so now the quiestion becomes, where does it get lost? Just out of curiosity, I plugged the keyboard to a Windows machine and used KeyboardStateView utility to read scancodes there. To my surprise, I'm still getting the same scancodes for keys pressed with and without Fn! So barring my complete misunderstanding of the process, it seems that either the keyboard itself is not doing what it supposed to do (now I'm curios whether Fn key is even wired to anything inside the keyboard) or it requires some custom (firmware? driver?) configuration to start behaving properly.

Elecrow-RD, 2020-11-20:
When we designed the keyboard, we only considered adding a key through FN to disable the touchpad, so other functions were not added. 
However, thank you so much for your suggestion, we will consider it in the next version of CrowPi2.
Thanks again!

closed 2020-11-20 by Elecrow-RD