Elecrow-RD/CrowPi2 #8: Fail to boot into X Window System desktop after resizing the swapfile

Pearl-852, 2020-11-21:
To reproduce:
1. Open a Terminal.
2. Enter the command: **sudo nano /etc/dphys-swapfile**
3. Uncomment the lines:
    - **CONF_SWAPFILE=/var/swap**
    - **CONF_SWAPFACTOR=2**
4. Comment the line:
    - **CONF_SWAPSIZE=100**
5. **Ctrl-X**, **Y**, and **Enter** to save the changes and exit to the command-prompt.
6. Invoke these three commands:
    - **sudo dphys-swapfile swapoff**
    - **sudo dphys-swapfile setup**
    - **sudo dphys-swapfile swapon**
7. Reboot the system.

After reboot, you will be in the cli only. Even entering **startx** will not start the X Window System.

Pearl-852, 2020-11-25:
FYI, just flashed a new system SDCard from your released image file, **crowpi2_release_v1.1.0_en.zip**, and the same symptom still exists.

_**Note**_: Tested the lastest official RaspiOS image, **Release date: August 20th 2020**, and there is no problem at all.

Elecrow-RD, 2021-01-04:
@Pearl-852 Please try to follow this link to resize the swapfile:
http://www.bitpi.co/2015/02/11/how-to-change-raspberry-pis-swapfile-size-on-rasbian/

Elecrow-RD, 2021-01-04:
@Pearl-852 Please follow the steps below to resize the swapfile:
STEP 1: Create swapfile and define it size
````
mkdir /swap 
cd /swap 
sudo dd if=/dev/zero of=swapfile bs=1024 count=2000000 
sudo mkswap swapfile
````

STEP 2: Active it:
````
sudo swapon swapfile
````

STEP 3: Modify the file "sudo nano /etc/fstab" by adding the following line:
````
/swap/swapfile  /swap  swap  defaults  0  0
````

Pearl-852, 2021-01-04:
@Elecrow-RD Thanks for your reply and instructions on how to create a separate swapfile. After my original report, I have been using a 2GB _swap partition_ on a USB thumbdrive already. So, I am okay running my CrowPi2 with Pi3B+ installed without it freezing up when the system ran out of physical memory.

The issue I was reporting is why running your CrowPi2 system image would fail to boot into X Windows after resizing the dphys-swapfile, while do the same using the official Raspi OS system image has no problem at all.

I would appreciate if you can explain the differences and/or exact cause of this problem for my own learning experience. Thanks

Pearl-852, 2021-04-05:
Just flashed a new system SDCard with the latest release image file, **crowpi2_release_v1.2.0_en_20210305.zip**, and the same symptom still exists.