Elecrow-RD/CrowPi2 #11: Plain Raspberry Pi OS does not work on CrowPi2

suranyami, 2020-12-01:
If I try to boot Raspberry Pi OS on the CrowPi2, it gets as far as the Desktop appearing then screen goes black and machine is unresponsive.

rabagliati, 2020-12-01:
Does it work with the switch (to the left of the fan) pushed towards the fan ?

That apparently disconnects all GPIO on the crowpi2 board.

Pearl-852, 2020-12-01:
Have you make the required changes to the config.txt file? See link below,
https://github.com/Elecrow-RD/CrowPi2#to-install-an-operating-system-eg-raspbian-retropie-from-scratch-follow-these-steps
I've flashed a new 64-bit RPiOS and made the required changes to the config.txt and it works fine. Othewise, CrowPi2 will shut itself off after about 10 seconds. That last line with **gpio=0=op,dl** is the fix.

suranyami, 2020-12-02:
Thanks, I'll give that a try. I guess I should've read the README more closely. :-(

closed 2020-12-11 by Elecrow-RD