Elecrow-RD/CrowPi2 #12: Where can we download or get source for fan, power management, brightness drivers?

suranyami, 2020-12-01:
In order to run plain Raspbian, we will need to be able to install drivers for the fan, power switch, brightness controls, etc.

Where is the source code?

Or, where is a repo for the Distro image?

rabagliati, 2020-12-01:
I think the fan is just 'on'.

The power switch schematics would be very helpful. SDA.0 (physical pin 27) has something to do with it - if you set it to 1 power is turned off after a few seconds.

Source code for all the projects is in /usr/share/code/project on the SSD.

Elecrow-RD, 2020-12-11:
> In order to run plain Raspbian, we will need to be able to install drivers for the fan, power switch, brightness controls, etc.
> 
> Where is the source code?
> 
> Or, where is a repo for the Distro image?

You don't need to install the drivers for them when running a standard Raspbian. You just need to add some commands in config.txt to make it work.

We haven't created a repo for building from a fresh Raspbian right now, Let me see what will happen later.

closed 2020-12-11 by Elecrow-RD