Elecrow-RD/CrowPi2 #14: Is there a recommended battery pack that will fit in the storage area?

suranyami, 2020-12-01:


Pearl-852, 2020-12-01:
The dimensions of the power bank drawer is 
- 147 x 78 x 9.5 mm or 
- 5.787 x 3.071 x 0.374 in

The Power Bank that is being shipped from Elecrow has the following spec
- Model:          LeTV LeUPB-501T
- Capacity:      5000/5090mAh 19.25/19.59Wh (MIN/TYP)
- Input:            5V-3A, 9V-1.9A, 12V-1.5A (MAX)
- Output:         5v-3A, 9V-2.3A, 12V-1.6A (MAX)
- Dimensions: 146.5 x 70.5 x 8.3mm
- Weight:         131g

suranyami, 2020-12-02:
Do you have a link to where we can buy that?

It's not listed on the Elecrow store and googling the model returns nothing.

I also didn't see an option to add a power bank to the Kickstarter pledge.

suranyami, 2020-12-02:
Correction: I can find links to buy the power bank, but they're only available from suppliers in Pakistan, India or Russia. Are there any suppliers in China or HK or AU?

Pearl-852, 2020-12-02:
Here is a link for the power bank, but it is in chinese
https://item.taobao.com/item.htm?spm=a230r.1.14.16.30fa1bdf3YNDKd&id=599877015547&ns=1&abbucket=12#detail

suranyami, 2020-12-03:
Thanks very much! Will order the one from TaoBao.

suranyami, 2020-12-03:
Also found this one on Aliexpress:

https://www.aliexpress.com/item/32825723715.html?spm=a2g0o.productlist.0.0.323c2772eE1pIn&algo_pvid=5936ed21-6fda-4057-9f07-079d1968394c&algo_expid=5936ed21-6fda-4057-9f07-079d1968394c-8&btsid=0bb0624316069570164586842e5432&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

closed 2020-12-11 by Elecrow-RD

khalwat, 2021-12-26:
For anyone finding this thread... it took me a while to find a power bank that was thin enough to fit... this one fits the specs:

https://www.amazon.com/gp/product/B07W5F556K/

MichaelDimmitt, 2023-12-08:
This is one I found, not sure how it will do yet since it is 5v 2.1a instead of 5v 3a but will report back after trying.
https://www.amazon.com/Attom-Tech-Portable-External-Emergency/dp/B07JZCBXMM