Elecrow-RD/CrowPi2 #18: "Low Voltage Warning" indication when powering through the 5V USB Power Supply Port

Pearl-852, 2020-12-06:
The Lightning Bolt (Low voltage warning) icon appears constantly when powering the CrowPi2 through a 5V/3A Power Bank or Power Adapter. 

Below are some voltage measurement comparisions that I took,
1. Using the included 12V/2A Adapter, 5V@GPIO = **5.22V**
2. Using a 5V/3A Power Bank, the input voltage = 5.12V and the 5V@GPIO = **4.74V**

rabagliati, 2020-12-06:
You need a short, thick (high current) USB cable between the power bank and Crowpi2.

At 12V, my Crowpi2 draws 0.54 Amps. At 5V, it will draw over 1.3 amps.

My suggestion:- Just use the 12V barrel connector.

Elecrow-RD, 2020-12-11:
> The Lightning Bolt (Low voltage warning) icon appears constantly when powering the CrowPi2 through a 5V/3A Power Bank or Power Adapter.
>
> Below are some voltage measurement comparisions that I took, [...]

Yes, the low voltage warning icon will appear when using the power bank to power the CrowPi2. This is related to the relatively large power consumption of the Raspberry Pi 4 and the many sensors on the board, but this is a normal phenomenon and will not affect the normal trial.

closed 2020-12-11 by Elecrow-RD

Pearl-852, 2020-12-11:
I cannot imagine with your reply that "This is a normal phenomenon and will not affect the normal trial."

Here are the official documentation from Raspberry Pi Foundation on this subject.

> **Power Supply Warnings**
On all models of Raspberry Pi since the Raspberry Pi B+ (2014) except the Zero range, there is low-voltage detection circuitry that will detect if the supply voltage drops below 4.63V (+/- 5%). This will result in a warning icon being displayed on all attached displays and an entry being added to the kernel log.

> If you are seeing warnings, you should improve the power supply and/or cable, as low power can cause problems with corruption of SD cards, or erratic behaviour of the Pi itself; for example, unexplained crashes.

_Reference link:_ [https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md](url)

So, I cannot agree that "This is normal and will not affect the normal usage."

In my opinion, and sorry to say, "This is a simple case of failed engineering design." Because this symptom appears constantly and not intermittently. And these warning instances are all being logged in the Kernel log. I cannot believe this symptom was not observed during your product development.

With all said, I really liked the CrowPi2 concept and hope the future model is better improved.