Elecrow-RD/CrowPi2 #20: LeTV LeUPB-501T Power Bank User Guide

Pearl-852, 2020-12-14:
The CrowPi2 advanced and deluxe kit includes a 5V/3A power bank, Model: LeTV LeUPB-501T. But I believe the user manual was not included. So, I have taken the liberty recreated the power bank user manual found from an online video at

[https://www.bilibili.com/s/video/BV1MJ411G7CD](https://www.bilibili.com/s/video/BV1MJ411G7CD)

The manual was originally written in Chinese only and I have translated it into English also. If you think it would be helpful for others, please feel free to post the attached pdf file on your official website.

[LeUPB-501T User Manual.pdf](https://github.com/Elecrow-RD/CrowPi2/files/5689606/LeUPB-501T.User.Manual.pdf)

716green, 2021-01-01:
According to your translated guide (thank you by the way, that was awesome of you) - it appears that I have a full charge, 4 solid bars. It doesn't seem able to power the laptop. Is there something I'm missing here? Is there a discharge mode? Pressing the button doesn't seem to do anything.

Pearl-852, 2021-01-01:
@716green According to the original user guide, the power bank will power on automatically into the discharge mode when a compatible device (using a USB-C connection) is connected. However, CrowPi2 uses a micro USB connection, thus mean it is not compatible for the power bank to auto power on. So, you will need to manually press the Power button on the power bank first, then press the Power button on the CrowPi2 to boot, which you've already found out

When the power bank is connected to a USB charger, then it will goes into the charge mode automatically.

The way to tell the power bank is functioning in the charge or discharge mode is by the status of the 4 LEDs indicator, as shown in the table on page 4.

- While charging, at least 1 LED will be blinking until it is fully charged (All LEDs On solid).
- While discharging, none of the LED should blinks until the power bank capacity drop below 5%.

Hope my own understanding and explaining is clear to you.

716green, 2021-01-01:
Very clear. Thank you. This is what I needed to hear.

Pearl-852, 2021-03-19:
The User Guide is now available in the FAQ section. See [Here](https://github.com/Elecrow-RD/CrowPi2/blob/main/faq/PowerBankUG.md)

closed 2021-03-19 by Pearl-852