Elecrow-RD/CrowPi2 #21: Screen resolution or bigger fonts in software

Karol3500, 2020-12-26:
Hi,

I'm using the official image. While the screen works great in the default resolution, some of the elements are terribly hard to use at that resolution. For example, Scratch has a script pane that's not resizeable and the fonts are very small. I have a good sight, but after a while my eyes get really fatigued when using the CrowPi2.

I've tried changing resolution to 1280×720. The screen looked less sharp than with full-HD, but it was also more readable. However this way it's impossible to follow the lessons, as the CropPi2 software (the home screen, the lessons window) seems to be completely neither scalable nor resizeable.

Are there some adjustments that can be made (other than a magnifying glass software) so that the software on the screen would be easier for eyes (simply, bigger)?

Best regards,
Karol

Pearl-852, 2020-12-28:
There are 3 major version of Scratch apps, each behave differently as below
### Scratch: 
1. Open the Terminal app.
2. At the command prompt, type _`nano ~/.scratch.ini`_ and press **Enter**.
3. Add the line, _`fontscale=1.3`_ (Experiment with different value as desire)
4. Press **Ctrl**+**X**, **Y** to save the changes.

### Scratch 2:
1. In the Scratch Desktop, hold **Shift** and click on the **Globe** icon.
2. Click on **set font size** and select your desired size (ex. 13).
    _**Note**: This set the font size of the Blocks only._

### Scratch 3:
I'm afraid, there is no solution or workaround available yet that I'm aware of.

premsatha, 2020-12-29:
Hi,

I hope someone is looking into this issue.The font is so tiny. I hope we have a solution soon.

Pappy2670, 2020-12-30:
I also agree the Scratch font is virtually unreadable.  Will try the solution Pearl-852 supplied.

Pearl-852, 2020-12-31:
Okay, I have a new workaround.

By using **Pixel Doubling** to make fonts more readable (larger) on small screen with high resolution, however, there are drawbacks
* The _CrowPi2 menu software_ refuses to  run, complaining that "**The CrowPi2 software needs to run at 1920x1080 resolution!**" _`(If Elecrow would fix this issue with the Pixel Doubling enabled, then it could be useful as a workaround solution)`_.
* **Pixel Doubling** affects the entire system. Enabling pixel doubling simply draws every pixel in the desktop as a 2×2 block of pixels on the screen, making everything exactly twice the size and resulting in a more usable desktop. But this will sacrifice screen quality.
* Will need more fiddling with other appearance settings to suit your own liking.

**To enable Pixel Doubling:**
1. Click on Menu (Raspberry icon).
2. Select **Preferences** and click **Raspberry Pi Configuration**.
3. The **Raspberry Pi Configuration** window appears.
4. On the **Display** tab, click on **Pixel Doubling: ( ) Enable**.
5. Click **OK** and reboot the system.
_**Note:** After reboot, the CrowPi2 menu software will refuse to run, and everything on the screen becomes very big. You will need to customize appearance settings as below._

**To customize Appearance Settings:**
1. Click on Menu (Raspberry icon).
2. Select **Preferences** and click **Appearance Settings**.
3. The **Appearance Settings** window appears.
4. On the **Menu Bar** tab, set **Size: Medium (24x24)**.
5. On the **System** tab, set **Font** to _PibotoLt 8_ and **Mouse Cursor**: _Small_.
6. Finally, click **OK** and that's it.
_**Note:** You may wants to experiment with different appearance settings to suit your own liking._

Hope these instructions help. Please feel free to comment/post your appearance settings to help others.

Pappy2670, 2020-12-31:
@Pearl-852 thanks for posting this.
Pixel Doubling worked for me.  I used Small Menu Bar.
Wonder if there is a way to access the CrowPi2 lessons now that I can read the screen.

Pearl-852, 2021-01-01:
@Pappy2670 All the CrowPi2 lessons are stored locally in the ` /usr/share/.user/course ` directory, then under its respective sub-directories for different applications accordingly (i.e. python, scatch). The ` ./zh ` sub-directory contains the same lessons for Chinese users.
### Directory tree structure:
[image]
Notice that the ` .../.user/... ` directory is a hidden directory, so you will need to select **Show Hidden** in the file manager in order to view it.

ontheslab, 2021-01-04:
Very much hope they are working on this! Pixel Doubling is almost a requirement on small screens!

ontheslab, 2021-01-04:
I would like to add my support to what @Pearl-852 is saying. To be useable to a very large number of students - including this half blind teacher - such an option on the tiny screen would seem to be a requirement.

BoneRoller, 2021-01-05:
I mentioned this tiny font problem (which is a BIG problem) in the Kickstarter comments a while back and haven't seen Elecrow respond. But as others mentioned, this issue makes the Scratch environment nearly unusable with the built-in screen.

ontheslab, 2021-01-05:
Link back to Kickstarter comments for reference:
https://www.kickstarter.com/projects/elecrow/crowpi2-steam-education-platformand-raspberry-pi-laptop/comments?comment=Q29tbWVudC0zMDg3NTM2MA%3D%3D

Karol3500, 2021-01-14:
Thanks Pearl-852 for the suggestions, looks promising. I've left the device with my nephew and will be able to test those next week when I'll return to home.

If I understand pixel doubling correctly, the only problem that prevents us from using the crowpi software with reasonable font size is a lack of UI scalability (I've managed to run it in different screen resolution and the crowpi app was only partially visible in the screen).
And for Scratch, only Scratch 3 doesn't seem to have any workaround for scaling.

Pearl-852, 2021-04-05:
The problem that Crowpi2 software refuses to run when **Pixel Doubling** is enabled has been confirmed and was fixed on V1.2.0 system image. Thus, the tiny fonts issue when using **Scratch 3** can be adjusted using **Pixel Doubling** as a workaround solution.

@Karol3500 
Please try the V1.2.0 [system image](https://github.com/Elecrow-RD/CrowPi2#official-crowpi2-system-image). If you are satisfied with the result, then, please close this issue.