Elecrow-RD/CrowPi2 #22: The CrowPi2 menu software refuses to run when Pixel Doubling is enabled.

Pearl-852, 2020-12-31:
The _CrowPi2 menu software_ refuses to run, complaining that "`The Crowpi2 software needs to run at 1920x1080 resolution!`".

### To duplicate:
1. Click on Menu (Raspberry icon).
2. Select **Preferences** and click **Raspberry Pi Configuration**.
3. The **Raspberry Pi Configuration** window appears.
4. On the **Display** tab, click on **Pixel Doubling: ( ) Enable**.
5. Click **OK** and reboot the system.

**Note:** This issue affects the usefulness of a workaround solution for [Issue #21 Screen resolution or bigger fonts in software](https://github.com/Elecrow-RD/CrowPi2/issues/21#issuecomment-752980860).

Elecrow-RD, 2021-01-04:
@Pearl-852 
The CrowPi2 software only can be run at 1920x1080 resolution! If you set other resolutions, the software will not be able to adapt.

closed 2021-01-04 by Elecrow-RD

Pearl-852, 2021-01-04:
@Elecrow-RD Seeing that particular display message, it is quite obvious that the Crowpi2 software is written to run at 1920 x 1080 resolution only without your reply. As I have indicated on the bottom notes, this is related to the issue #21 that requires to enable **Pixel Doubling**.

Please also note that you are marketing the CrowPi2 as a programming learning platform for general users, and not able to run the Crowpi2 menu software when Pixel Doubling is enabled is extremely difficult for them to learn. For the more advanced users, it would be okay to load your learning lessons  manually without the Crowpi2 menu.

So, I recommend you to look into modifying your Crowpi2 software to be able to run with **Pixel Doubling** enabled, unless you can come up with an alternative workaround solution for the issue #21. Thanks.

ontheslab, 2021-01-04:
I would like to add my support to what @Pearl-852 is saying. To be useable to a very large number of students - including this half blind teacher - such an option on the tiny screen would seem to be a requirement.

Pearl-852, 2021-01-06:
If you will so agreed to modify the Crowpi2 software, I have one other modification request, adding a way to quit the software gracefully (ex. **Ctrl**+**Q**). 

Currently, I've created a simple bash script on the desktop to forcefully kill the Crowpi2 process as shown below,

```
#!/bin/bash
#=====================================
# Name: Kill_Crowpi2.sh
# Purpose: Kill the process 'Crowpi2'
#=====================================

killall Crowpi2
exit 0
```

Pearl-852, 2021-04-05:
@Elecrow-RD 
The problem that Crowpi2 software refuses to run when **Pixel Doubling** is enabled has been confirmed and was fixed on V1.2.0 image. Thanks :)