Elecrow-RD/CrowPi2 #23: CrowPi2 shuts down when starting RetroPie

Gamesman, 2021-01-02:
When I first got the CrowPi2 I was able to play RetroPie. After going through some of the programming tutorials on the system card, I can not load the RetroPie card without the CrowPi2 shutting down. I swapped out the card and put the system one back in and it starts fine. What happened to cause this? I have even tried to use new cards with fresh images of RetoPie on them and it still shuts down. I have tried with the Sensor Switch in both the A and B position.

https://user-images.githubusercontent.com/2320809/103464878-60851e00-4cf4-11eb-9b27-7ae147c038b1.mp4

Pearl-852, 2021-01-03:
@Gamesman  Go to [this link](https://github.com/Elecrow-RD/CrowPi2#to-install-an-operating-system-eg-raspbian-retropie-from-scratch-follow-these-steps), and follow the instruction in step 2 to make sure those lines of command are present. If not, add them in.

The key solution for your issue is the last line of those commands `gpio=0=op,dl`

Elecrow-RD, 2021-01-04:
@Gamesman As "Pearl-852" said, you need to add some commands in your config.txt file to fix this problem:
    hdmi_force_hotplug=1
    max_usb_current=1
    hdmi_group=2
    hdmi_mode=87
    hdmi_cvt 1920 1080 60 6 0 0 0
    hdmi_drive=2
    enable_uart=1
    gpio=0=op,dl

Save the file and eject SD card, plug it into CrowPi2 and boot.

Besides, there are many errors that occur when you boot the image, which means it might be damaged, please download the latest Retropie image(https://github.com/RetroPie/RetroPie-Setup/releases/download/4.7.1/retropie-buster-4.7.1-rpi4_400.img.gz) and write it into your SD card and then add the above commands.

closed 2021-01-04 by Elecrow-RD

Pearl-852, 2021-01-04:
As a side note on **max_usb_current**. This command has been deprecated and has no effect anymore. 

See official documentations from Raspberry Pi Foundation at [this link](https://www.raspberrypi.org/documentation/configuration/config-txt/misc.md).