Elecrow-RD/CrowPi2 #26: Schematics for Sensor switch and power supply

rabagliati, 2021-01-06:
I am enjoying my CrowPi2 - and I am glad that libraries for access to all the sensors are available.

I am using my Crowpi2 to interface to other hardware, at the same time as driving the LED and LCD displays - it makes an excellent controller.

Please - can I echo Pearl-852 in asking for release of the schematics around the Sensor switch, and the power supply switching between Barrel jack and USB power. The Crowpi2 is a Makers ideal platform, and in that spirit can I make a plea to release those schematics.

Pearl-852, 2021-01-08:
I would love to see the schematics of the USB power port circuit, as it is not delivering enough voltage to the Raspberry Pi causing the Pi to display the "**Low Voltage Warning**" message. (Issue #18)

rabagliati, 2021-01-11:
This is another design, faced with the same problems around packaging as the CrowPi2 designers.

YouTube series - 
https://www.youtube.com/channel/UCnmjLV ... aprpLuw2LQ

There are pictures at https://imgur.com/a/BkcIFII

also he puts in a power booster. For people here, I think the best is to run it off the barrel jack.

CrowPi2 designers put most of the power components in the unit - so, use them - from 12V.

For me it would be more convenient from 18V - my power drill battery - or even 19V - the inevitable old laptop charger doing nothing. It is for this reason I would like the schematics.

berista, 2021-09-05:
Hello everyone
I want to run my Crowpi2 on a brand new Powerbank and was running into some strange observations.
While on the barrel jack, i measured 5.11V, on the Powerbank 4.8ish Volts. I tested with a different powerback with fixed 5V output
I disconnected the 2 phase connector cable from the Rpi and plugged the powerbank back in. 
Obviously The RPi is also powered via GPIO Header (also giving low power lightning bolt) but i have no measurement atm.

Tested on Rpi 4, 4GB, Rev1.4

It is mentioned on the official forum, that undervoltage warning is by "design" (Downward compatibility for Rpi 3 with micro USB vs. USB-C on RPi4). It somehow loses 0.3 Volts in the internal wiring. 
I hope this helps