Elecrow-RD/CrowPi2 #27: Crowpi2 software improvement request

Pearl-852, 2021-01-07:
Just trying to list all the improvement requests in a single place.

**Request 1.** Modify the Crowpi2 software to be able to run in any resolution (Issue #22)

Pearl-852, 2021-01-07:
**Request 2.** Add a way to quit the Crowpi2 software gracefully (ex. **Ctrl**+**Q**). This request is related to the reported issue #24.

Currently, I've created a simple bash script on the desktop as a workaround to forcefully kill the Crowpi2 process as shown below,
```
#!/bin/bash
#=====================================
# Name: Kill_Crowpi2.sh
# Purpose: Kill the process 'Crowpi2'
#=====================================

killall Crowpi2
exit 0
```

Pearl-852, 2021-01-07:
**Request 3.** Rename the popup comments of the Crowpi2 icon on the Task Bar to a more intuitive name (ex. **_Crowpi2 programming environment_**)

[image]

ontheslab, 2021-01-08:
Great idea, I see they still have #22 Closed, I do hope its all considered in with #21 - Planning for student work in our (Australian) new school year and this screen size issue really needs addressing!

Elecrow-RD, 2021-02-18:
@Pearl-852 
Thank you so much for your feedback and suggestions. We are working on it to fix all the requests you mentioned. I believe all of them will be resolved soon!

Pearl-852, 2021-02-19:
@Elecrow-RD Good to hear. Will add a few more suggestions later.

Pearl-852, 2021-02-23:
## Login screen:
**A**. The **LOGIN** button is not part of the keyboard scrolling action group (pressing the Tab/Shift+Tab). Thus, in order to activate LOGIN, the Track-Pad or Mouse must be used to navigate the mouse pointer over the **LOGIN** button and then click on it.
- **Current action with the Tab key input:**
USERNAME > PASSWORD > Create Account > Forget Password?
- **Suggestion:**
USERNAME > PASSWORD > Create Account > Forget Password? > LOGIN

**B**. The **Enter** key has no function at all when pressed
- **Current**
No action.
- **Suggestion:**
Works as you have clicked on the **LOGIN** button.

**C**. After a successful login when the Login function was invoked from clicked on **Learning** in the main screen,
- **Current**
Returns to the main screen. Thus require clicking on **Learning** again.
- **Suggestion:**
Goes into the Learning lessons selection screen for Python or Scratch.

Pearl-852, 2021-03-06:
**Request 5.**
A bug that was reported under issue #28

Pearl-852, 2021-03-09:
**Request 6.**
When the Crowpi2 software starts but the Sensor Switch is not set on the **A** position, the pop-up warning message is not intuitive enough to be understood.

- **Current:**  
[image]

- **Suggestion:**  
[image]

Pearl-852, 2021-03-10:
**Request 7.**
An arbitrary file naming, **1.desktop**, see issue #25

Pearl-852, 2021-03-10:
**Request 8.**
Typo found in the ***/usr/share/applications/crowpi2.desktop*** file.

- **Current:**
```
[Desktop Entry]
Type=Application
Version=1.0
Name=Crowpi2
GenericName=Crowpi2
Comment=Crowpi2 programming environment
Exec=Crowpi2
NotShowIn=GMOME;XFCE;     <-- Typo, should be "GNOME"
Icon=/usr/share/elecrow/picture/Crowpi2.png
StartupNotify=true
Categories=Education
```
- **Suggestion:**
```
[Desktop Entry]
Type=Application
Version=1.0
Name=Crowpi2
GenericName=Crowpi2
Comment=Crowpi2 programming environment
Exec=Crowpi2
NotShowIn=GNOME;XFCE;
Icon=/usr/share/elecrow/picture/Crowpi2.png
StartupNotify=true
Categories=Education
```

Pearl-852, 2021-03-10:
**Request 9.**
Add 2 features to the Crowpi2 software menu screen:  
1. Shows software version.  
2. Provide a way to quit the software.  *(Relates to **Request 2** above)*  

[image]

Elecrow-RD, 2021-03-17:
We have resolved some of the issues raised earlier and updated to the latest mirror. I have updated the latest image link to your Pull Request. For improvement issues raised later, we will continue to optimize in the next version of the image. Thanks~

Pearl-852, 2021-03-17:
Great! Is there a release notes for the new image version?

Pearl-852, 2021-04-04:
**Request 10**
Revised the English text files (Total=28 files) that explain each built-in module in the `/etc/.user/Text/en/` directory to make them more readable. All revised files are included in the [Text.zip](https://github.com/Elecrow-RD/CrowPi2/files/6254806/Text.zip)

Changes summary are in the [TextDirContents.docx](https://github.com/Elecrow-RD/CrowPi2/files/6254798/TextDirContents.docx)

Pearl-852, 2021-04-05:
**Request 10**
Improved the English text files (Total=28 files) that explain each on-board module for Python lessons in the `/usr/share/.user/Text/en/` directory to make them more readable. All revised files are included in the [Text.zip](https://github.com/Elecrow-RD/CrowPi2/files/6254806/Text.zip)

Changes summary are in the [TextDirContents.docx](https://github.com/Elecrow-RD/CrowPi2/files/6254798/TextDirContents.docx)

Pearl-852, 2021-04-05:
**Request 11**
There is an old config file `/boot/config1.txt` included.

**Suggestion:**
Delete it before creating the system image.

Pearl-852, 2021-04-05:
**Request 12**
Some suggestions to the config file contents `/boot/config.txt` as follows,

1. A good practice when defining a custom filter section, i.e. `[EDID=RTK-RTK_FHD]` is to reset it with a `[all]` line. 
2. It is not neccessary to define a custom mode `hdmi_mode=87` for the CrowPi2 LCD screen. Use 'hdmi_mode=82` would have the same settings.
3. The `hdmi_cvt 1920 1080 60 6` parameter value defined for the aspect ratio, **6** (15:9), is incorrect. The correct value **3** (16:9) should be used.
- **Current:**
```
[EDID=RTK-RTK_FHD]  
enable_uart=1  
hdmi_force_hotplug=1  
hdmi_group=2  
hdmi_mode=87  
hdmi_cvt 1920 1080 60 6  
hdmi_drive=2  
gpio=0=op,dl  
```
- **Suggestion 1:**
```
[EDID=RTK-RTK_FHD]  
enable_uart=1  
hdmi_force_hotplug=1  
hdmi_group=2  
hdmi_mode=82  
hdmi_drive=2  
gpio=0=op,dl  
[all]
```
- **Suggestion 2:**
```
[EDID=RTK-RTK_FHD]  
enable_uart=1  
hdmi_force_hotplug=1  
hdmi_group=2  
hdmi_mode=87  
hdmi_cvt=1920 1080 60 3  
hdmi_drive=2  
gpio=0=op,dl  
[all]
```

Pearl-852, 2021-04-05:
**Request 13**
The `/etc/wpa_supplicant/wpa_supplicant.conf` file contains 7 Elecrow SSID
   1) Elecrow888
   2) Tony
   3) Elecrow
   4) ChinaNet-elecrow2
   5) RM2100_GUEST_7739
   6) ELECROW
   7) 123456

**Suggestion:**
Delete it before creating the system image.

Pearl-852, 2021-04-05:
**Request 14**
The `/home/pi/.bash_history` file still contains few lines of contents. It is a lot better than the previous version.
```
rm .bash_history 
df -h
vim /boot/config.txt 
sudo vim /boot/config.txt 
reboot
sudo vim /boot/config.txt 
reboot
```
**Suggestion:**
Delete it before creating the system image.

Pearl-852, 2021-04-05:
**Request 15**
The file `/home/pi/.asoundrc` - This is an old ALSA configuration file that is no longer being used starting December, 2020 of RasPiOS upgrade.

**Suggestion:**
Delete it before creating the system image.

Pearl-852, 2021-04-05:
**Request 16**
The file `/home/pi/.viminfo` - This viminfo file was auto generated by vim.

**Suggestion:**
Delete it before creating the system image.

Pearl-852, 2021-04-05:
**Request 17**
Some unnecessary files are found as shown below:
`/usr/share/.user/course/._.DS_Store`
`/usr/share/.user/course/.DS_Store`
`/usr/share/.user/course/en/python/Play_songs_using_the_IR_remote/需要手动修改HTML.txt` (0 byte)
`/usr/share/.user/course/en/python/using_the_DHT11_sensor/debug.log` (67 bytes)

Suggestion:
Delete them before creating the system image.

Pearl-852, 2021-04-05:
**Request 18**
User account - There is an existing account, **Crowpi2**, created
`/usr/share/.user/account/Crowpi2`

**Suggestion:**
Delete it before creating the system image.

Pearl-852, 2021-04-06:
**Request 19**
Chinese files - Not sure these files are neccessary (Same on V1.1.0)
`/usr/share/.user/font/mononoki/下载说明.htm`
`/usr/share/.user/font/mononoki/使用帮助(河东软件园).url`
`/usr/share/.user/font/mononoki/使用说明.txt`

**Suggestion:**
Delete them if not neccessary before creating the system image.

Pearl-852, 2021-04-06:
**Request 20**
Chinese directory/file naming - Even though they are specifically indicates under the `.../.user/Text/en/` English directory (Same on V1.1.0)
`/usr/share/.user/course/en/python/Play_songs_using_the_IR_remote/需要手动修改HTML.txt`
`/usr/share/.user/course/en/python/python课程顺序.xlsx`
`/usr/share/.user/course/en/python/Tracing_experiment/Picture/红外反射传感器.bmp`
`/usr/share/.user/font/mononoki/下载说明.htm`
`/usr/share/.user/font/mononoki/使用帮助(河东软件园).url`
`/usr/share/.user/font/mononoki/使用说明.txt`
`/usr/share/.user/html/en/face/Install_OpenCV_3/Pictures/图片1.png`
`/usr/share/.user/html/en/face/Overview/Pictures/CrowPi2_人脸识别01正确.gif`
`/usr/share/.user/html/en/face/Test_camera/Pictures/QQ图片20200506191949.png`
`/usr/share/.user/html/en/face/准备好的人脸识别目录/`
`/usr/share/.user/html/en/minecraft/Find_block/Pictures/捕获.PNG`
`/usr/share/.user/html/en/minecraft/Find_location/Pictures/图片1.png`
`/usr/share/.user/html/en/minecraft/Generate_house/Pictures/图片1.png`
`/usr/share/.user/html/en/minecraft/Generate_house/Pictures/图片2.png`
`/usr/share/.user/html/en/minecraft/Minecraft_API/Pictures/图片2.png`
`/usr/share/.user/html/en/minecraft/Run_game/Pictures/图片1.png`
`/usr/share/.user/html/en/minecraft/TNT_detector/Pictures/TIM图片20200509143011.png`
`/usr/share/.user/html/en/minecraft/Write_data/CrowPi2_我的世界_纸盒教程02~1.mp4`

Attachment: [DirTreeV120-chsFilename.txt](https://github.com/Elecrow-RD/CrowPi2/files/6262300/DirTreeV120-chsFilename.txt)

**Suggestion:**
All should be renamed in English and its original references in the source code of Crowpi2 software accordingly.

Pearl-852, 2021-04-06:
**Request 21**
Duplicated video files - (Same on V1.1.0)
`/usr/share/.user/html/minecraft_en.mp4 (24.3MB)`
`/usr/share/.user/html/en/minecraft_en.mp4 (24.3MB)`
`/usr/share/.user/html/en/minecraft/Write_data/CrowPi2_我的世界_纸盒教程02~1.mp4 (24.3MB)`

**Suggestion:**
Optimize the source code of Crowpi2 software to reference to a single video file and delete the duplicate files.

Pearl-852, 2021-04-06:
**Request 22**
Typo found in the Lesson06.txt - ... from 1 to 9 ... --> from 0 to 9 (Same on V1.1.0)
`/usr/share/.user/Text/en/Lesson06.txt`

_**Note:** New file contents has been provided in Request 10 above._

Pearl-852, 2021-04-06:
**Request 23**
**Create ID** dialog box wording
[image]

**Currect:** 
Create Successful!

**Suggestion:** 
New account created successfully!

Pearl-852, 2021-04-06:
**Request 24**
**Note** dialog box wording
[image]

**Suggestion:**
- Title: Save Changes
- Message body: Do you want to save the changes before exiting?
- Buttons: [**D**on't Save]   [**C**ancel]   [**S**ave]
_(Note: **Bolded** letter indicates underlined and to be used as the keyboard shortcuts)_