Elecrow-RD/CrowPi2 #28: Python editor zoom

ljwinkler, 2021-01-14:
Hi,
While doing python lessons, I pressed "Ctrl" + "-" keys and the font got smaller.
However, "Ctrl" + "+" doesn't increase the size.
Is there any other combination?
Tried holding different key combination with the "+" - no luck.
Same with using mouse scroll, usually Ctrl + scroll up increases fonts in many applications.

Thanks and regards,
LJ

Pearl-852, 2021-01-14:
The actual keys input for "Ctrl" + "+" are **Ctrl**+**Shift**+**+**, because the **+** sign on the keyboard is activated by **Shift**+**+**.

ljwinkler, 2021-01-14:
Hey, sorry, I should be clearer in my original message.
I know that Shift + "=" button makes the + sign but it didn't work.
This was the first thing I tried :)

Pearl-852, 2021-01-14:
Are you using Thonny? Mine works fine increasing and decreasing font size using **Ctrl**+**+** and **Ctrl**+**-** respectively.
Can you type the **+** sign in your python programming?
Also, using the right **Ctrl** key, does it work?

ljwinkler, 2021-01-14:
I'm using the Crowpi2 app.
The one when you do the "Learning" and you pick python or scratch environments.
The + button works fine on its own, I can type it, etc. The Ctrl + combination works everywhere in Raspbian but not in the Crowpi2 app. I've also connected an external keyboard to confirm it is not the case. (I know how to use a keyboard :))

This is what combinations do:
- Ctrl and -     decreases font
- Ctrl and +    types "="
- Ctrl and Shift and -     nothing
- Ctrl and Shift and +    types "+"

The good news is that it looks like it not a persistent setting, after rebooting the RPi, the font comes back to the default setting.

Pearl-852, 2021-01-14:
I just want to make sure I am looking at the same app as you. When I click on the Python button on the Crowpi2 app, it is actually launching Thonny Python IDE 3.3.0 with Python 3.7.3
On mine, changing the font size is persistent, whether I reboot the system or just exit and open Thonny again. So, something is different.

ljwinkler, 2021-01-14:
Sure, no worries. Thanks for your help.
In Thonny, the Zoom In/Out works fine (Ctrl - or Ctrl Shift +). The app I'm talking about is the one that starts automatically when you boot the RPi from the System SD card.

You can also start it by running it from Desktop menu: Education -> Crowpi2
Then click on the big "Learning" button (provide your details or create an account, no internet needed, it just creates a local folder under /home/pi/user/).
Pick "PYTHON" and you will see an editor on the left. And that's the window I can zoom out but not zoom in :)

[image]
[image]
[image]
[image]

Pearl-852, 2021-01-14:
Sorry for my mis-understanding. I haven't actually goes into the Learning part.

ljwinkler, 2021-01-14:
I was just browsing through the lessons to see what are they about.
I'm not going to use it for future python development.
And while typing I just accidently pressed the "Ctrl" + "-" combination and couldn't revert it back :)
Thonny is much better for 'normal' programming, but I was just curious about those lessons.
Thanks again.

Pearl-852, 2021-01-19:
@ljwinkler Just got around trying the CrowPi2 Learning lessons, and I concur with your finding.