Elecrow-RD/CrowPi2 #30: How to use all these sensors? What are the pin assignments, how is everythgin conencted?

ringmybell, 2021-01-22:
I have received the CrowPi2, but I can not find anywhere on the device any code or description of how to use these sensors and what they are connected to.

I really do not want to pull this apart and use a multimeter to figure it out. 

Can you please direct me to this information so I can use Python or Node-Red running on the Pi to interface with these sensors, the kind of information I am looking for is.....

1. What sensors are connected to what GPIO
2. What sensors are connected vi SPI and what GPIO is the CS pin
3. What sensors are connected via I2C and what is the device ID

I noticed that Arduino is included, so maybe these sensors are actually interfaced to an Arduino chip tht is then interfaced to the Raspberry pi??

Without this information this great little device is a bit useless for its intended purpose. 

Thanks

Pearl-852, 2021-01-23:
Answers are listed [here](https://github.com/Elecrow-RD/CrowPi2#list-of-crowpi2-on-board-modules).

ringmybell, 2021-01-23:
Okay, thankyou.

For the fan, I mean the fan has a high frequency that gets annoying after sometime - there is nothing wrong with the fan, it is just annoying.

Am I right that the fan is directly connected to power and will always run, it is not possible to turn it on or off based on CPU temperature? 

If the fan is directly connected then what power is it running off, is it pulling from the %v on the PI or from the 12V(?) main supply. 

Thanks

Pearl-852, 2021-01-23:
Fan issue is replied on issue #32, to keep things clean.