Elecrow-RD/CrowPi2 #32: Fan is noisy and annoying. Is the fan hardwired, or can we control when it is turned on or off?

ringmybell, 2021-01-22:
The fan is a bit noisy. 

Can the fan be controlled with some python script to only switch on when CPU temperature is above a certain limit?

So is the fan hard wired to 12v power and will always run? Of can it be controlled via a GPIO from the Raspberry Pi? If it can be controlled can you please advise which pin it is connected to. 

Thanks

Pearl-852, 2021-01-23:
> For the fan, I mean the fan has a high frequency that gets annoying after sometime - there is nothing wrong with the fan, it is just annoying.
> 
> Am I right that the fan is directly connected to power and will always run, it is not possible to turn it on or off based on CPU temperature?
> 
> If the fan is directly connected then what power is it running off, is it pulling from the 5v on the PI or from the 12V(?) main supply.
> 
[image]
The fan is Brushless DC 5V type. I believe it is directly powered from the CrowPi2 PCBA. If you wish to switch it on/off based the Pi CPU temperature, you will need to build a simple drive circuit to be controlled by a GPIO pin and a python script to read the cpu temperature and switch the fan on/off accordingly.
_**Note**: Do **NOT** connect the fan directly to a GPIO pin, otherwise, you will risk damaging the GPIO or worst the Pi itself._

Pearl-852, 2021-01-23:
Alternately, you can replace the fan with the official [Raspberry Pi 4 case fan](https://www.raspberrypi.org/blog/new-raspberry-pi-4-case-fan/) and utilize the RaspiOS built-in settings to control the fan.

Pearl-852, 2021-01-24:
Found a 3rd alternative idea.

While I have a chance to poke around the CrowPi2 learning lessons, there is an on-board **Relay module** driven by the **GPIO-29**. So, you can rewire the fan to it. Then write a python script that read the Pi cpu temperature and switch the fan on/off accordingly

ringmybell, 2021-01-24:
That is a great idea.

I was going to use a transistor and build a circuit on the breadboard but that relay idea will be super simple.

Thanks!

CaptCOBOL, 2021-03-26:
Sorry that this is not helpful at all, and I have seen SO many people report this same problem. But, my fan that I received is exactly the same as pictured above, but generates so little sound that I barely hear it, and certainly no sound when the keyboard is over it in the CrowPi2 case. Of course, I'm 67 years old, and listen to rock, trance, and techno music way too loud... so... Maybe ask Elecrow for a replacement?