Elecrow-RD/CrowPi2 #34: RGB matrix isn't working

Sramsram, 2021-02-05:
Touch sensor, keypad matrix, humidity sensor, lc display are working fine - any demo app with the RGB matrix doesn't work.

(using CP2 with RPi4  2GB).

berista, 2021-09-05:
Hey there. 
I ran into the same issue I think.
I changed from Rpi 4 4GB rev. 1.1 to rev 1.4 and all the projects including RGB matrix were broken. In some the LCD Display was flashing instead.
I opened the Project in Mu, and the Debugger said it was because of an unsupported hardware revision
Replacing the older Pi fixed everything.
You may check what hardware revision

I hope this can be fixed with a Software update.

berista, 2021-09-06:
The Tech Support from my local Pi-store had the solution:
It's just uninstalling and reinstalling the library for the RGB Matrix:

1.
sudo pip uninstall rpi_ws281x
2.
sudo pip3 uninstall rpi_ws281x
3.
sudo pip install rpi_ws281x
4.
sudo pip3 install rpi_ws281x

gausseil, 2024-03-12:
Well, I had the same problem, reinstalled the libraries. The problem is now half resolved, because the rgb 8x8 matrix works but only for 37 leds out of the 64. A bit more than half of them... I tried to reinstall again, but nothing better.