Elecrow-RD/CrowPi2 #35: Bent Keyboard?

MrRKernelPanic, 2021-02-15:
Did anyone else receive their unit with the keyboard slightly bowed?  Mine is an is virtually unusable, when typing the left arrow key kicks in and messes up your typing.  The keyboard seems to have some flex in it, but I don't want to bend it.  Any suggestions on a fix, how to flatten the board or where to get a replacement?

CaptCOBOL, 2021-03-26:
Yes. Pretty severe bend in mine, but it rarely mistypes, so I'm able to use it fairly well. I think I measured the bend from middle to side and it's at least 4mm! The primary problem with the bend for me is that the keyboard rocks as I'm typing when I have it sitting on a flat surface. It does occasionally insert a random character here and there, but I'm a two finger typer and watch what shows in the input box anyway. Would be nice if Elecrow could have had better quality control on the shape (flatness) of the keyboard. But when it's inside the case, I don't even notice the curve... Just the occasional random added characters.

MrRKernelPanic, 2021-03-30:
Keyboard had a massive wonk.  I contacted support an they very kindly asked for photos / evidence.  They then replaced the keyboard (and mouse).  I wish more Asian companies this level or response and customer support.