Elecrow-RD/CrowPi2 #36: Connecting a Pi HAT

mrsfixit, 2021-02-20:
I'd like to connect a Pi HAT (namely a TV HAT) to the CrowPi.

When I connect the HAT to a Pi Zero W (running [Raspberry Pi OS Lite](https://www.raspberrypi.org/software/operating-systems/)) I see the following in `dmesg`:
```
[   24.597507] dvbdev: DVB: registering new adapter (CXD2880)
```
... but no such message when I connect the same HAT to the PI HAT header on the CrowPi.

I'm assuming I need to flick the "switch" from `CONNECT SENSOR` to `DISCONNECT SENSOR`,
but is there something I need to be doing on the CrowPi to get the HAT to work?

Do I perhaps need to add or remove something from `/boot/config.txt`?

mrsfixit, 2021-02-21:
Ok, some progress...

I tried removing the Pi 4 from the CrowPi2 and plugging in my TV HAT directly:
[image]

And that *still* didn't work.

However when I removed the:
`gpio=0=op,dl` line for `/boot/config.txt` *then* everything finally started working.

Nb. I had tried removing that config line previously but it didn't work when the HAT was connected via the CrowPi2.

So is there perhaps something faulty with the connection between the CrowPi2 and my HAT?
Or, is it possible that the `DISCONNECT SENSOR` switch doesn't work as I'm expecting?

mrsfixit, 2021-02-21:
I thought maybe the pins on the headers I'm using were too short on one side.
[image]

However after searching online & in electronics stores I was unable to find headers that had length on both sides.
(As I'm assuming the headers I had were designed to have the short side soldered onto boards, so only needed to have length on one side).

So I looked up the [pinout of the tv hat](https://www.raspberrypi.org/forums/viewtopic.php?p=1483088#p1483088).
Then as a hack to get pins that were long enough, I ripped out the metal pins from the header with some pliers and inserted them directly in to the corresponding sockets on the CrowPi2 like so:
[image]

Then I attached the TV HAT to the CrowPi2 like so:
[image]

(Nb. using the same `/boot/config.txt` I did when I saw the TV HAT being detected correctly by the kernel when attached directly to the Pi 4)

However - **still** - the HAT is not being detected.

Any ideas on what I could try next?

kensuke, 2021-02-21:
long header on both sides is here in Japan store, but didn't shipped overseas.
https://akizukidenshi.com/catalog/g/gC-12180/

rabagliati, 2021-02-21:
May I leave another request for the schematic diagrams around the pins, the sensor switch, I2C and power be made available to help us with our projects.

mrsfixit, 2021-02-22:
Thanks @kensuke, although I would have thought my pin hack should have worked.
And besides the LED on my HAT always lit up - so *surely* the pins are long enough.
I'm just suspicious that something else is happening around the pins that's messing with my HAT.

Has anyone else ever got a HAT of any kind working plugged into the CrowPi2?
If so which ones?
And do any of them use I2C like my HAT does?

kensuke, 2021-02-23:
[image]
confirmed some devices
~~~~
CrowPi2 - long header pin - pHAT Stack with ribbon cable
 - Crowtail Base Shield
   - (I2C) Color Sensor - working
   - (Digital) LED(red) - working
   - (Analog) Pulse Sensor(green) - maybe working (I didn't confirm this time)
 - Scroll pHAT HD - working clock.py shows 09:35
 - XmasTree - working
 - Piano HAT - don't working. connect directry to CrowPi2 without pHAT Stack, also don't working... In my ambiguous memory, Piano HAT works with CrowPi2, so this is a software environment problem.
~~~~

- long header pin with both sides https://akizukidenshi.com/catalog/g/gC-12180/
- pHAT Stack https://shop.pimoroni.com/products/phat-stack?variant=658973392906
- Crowtail- Base Shield for Raspberry Pi 2.0 https://www.elecrow.com/crowtail-base-shield-for-raspberry-pi-p-1532.html
  - (I2C) Color Sensor https://www.elecrow.com/wiki/index.php?title=Crowtail-_I2C_Color_Sensor
  - (Digital) LED https://www.elecrow.com/wiki/index.php?title=Crowtail-_LED
  - (Analog) Pulse Sensor https://www.elecrow.com/wiki/index.php?title=Crowtail-_Pulse_Sensor
- Scroll pHAT HD – White https://shop.pimoroni.com/products/scroll-phat-hd?variant=38472781450
- XmasTree http://www.kibanhonpo.com/lab/xmastree-demo/
- Piano HAT https://shop.pimoroni.com/products/piano-hat

mrsfixit, 2021-02-23:
Oh wow - thanks once more @kensuke, your ✨amazing✨ collection has given me some hope.

Ahhh I see long & short header pins in your picture.
So are you able to confirm that when you put the long pins into your ribbon cable & the short side into the CrowPi2 that the connection doesn't work?  (And that switching that out for the long pins on both sides works?)
(That is *really* useful to know.)

Yer my local electronics store didn't have the double sided long header pins, but I'll try ordering some online and try again.

BTW (if anyone who stumbles across this thread is interested), I found the pinouts for a couple of your HATs here:
 * https://pinout.xyz/pinout/scroll_phat_hd
 * https://pinout.xyz/pinout/piano_hat

mrsfixit, 2021-02-23:
(see also my attempt to document the pinout for the TV HAT - https://github.com/Gadgetoid/Pinout.xyz/pull/406)

mrsfixit, 2021-03-05:
@kensuke - I was just wondering if you could please confirm that the difference between this working & not working is due to having the long pins on *both* sides?

mrsfixit, 2021-03-31:
FYI, to anyone interested, I got some headers - long pins on both sides - but still the [TV HAT](https://www.hackster.io/news/introducing-the-raspberry-pi-tv-hat-7992512c6195) was *not* detected when plugged in via the CrowPi2.