Elecrow-RD/CrowPi2 #37: 12V to 5V conversion Issue

Gaalua, 2021-02-22:
Hi,

I've tried to use the retropie sd card, and I've used it without any problem a few times, but now the crowpi2 shuts down during the boot sequence while being powered with the included 12V PSU.

I've tried to troubleshoot the issue and here the steps I've followed.

I've swapped the sd cards and it was able to boot to the crowpi os. After swapping again, no boot.

I've reflashed the retropie sd card, same problem.

Then I've disconnected the cable labeled "Power Cable" from the board and plugged in an official RPi PSU and I was able to boot in retropie and play without any issue!

I wanted to check if the 12V PSU that is included was at falt so I reconnected everything and swapped the PSU but still, the shutdown would occur.

It really looks like the 12V to 5V converter that is on the board is not working properly and to make matter worse, it seems to be degrading since I could boot in retropie at the beginning but not anymore.

Looking around online, I noticed that other people were complaining about the "low voltage" warning when connected to the power bank. And, since I had the same warning while connecting it to the right side of the machine, I was curious to see if there really was an issue with it. I connected the battery directly to the RPi the way mentioned previously:
It worked flawlessly and didn't even show the the low voltage warning!

It really looks like the problem comes from the conversion from 12 to 5V.

Could you help me with this? I didn't see any screw so I couldn't open the machine to see if there was any damaged component.

PS: I've posted a similar comment on the kickstarter page before I found this github. I saw that a similar issue was raised here: https://github.com/Elecrow-RD/CrowPi2/issues/23 although in my case it did work both at the beginning and after connecting power directly to the RPi. For me, this is a strong hint that the origin of the problem is not software based.

Pearl-852, 2021-02-22:
This is a known symptom. Just add the following statement to the bottom of the `/boot/config.txt` file in your Retropie SD Card.

`gpio=0=op,dl`

Gaalua, 2021-02-22:
Ok, thanks, It was indeed software.

closed 2021-02-22 by Gaalua