Elecrow-RD/CrowPi2 #41: Sound jack for headphones does not work after dist-upgrade

Dirk1966, 2021-03-04:
Instlled image --> sound worked on headphone jack
"sudo apt-get update && sudo apt-get upgrde", reboot --> sound worked
"sudo apt-get dist-update && sudo apt-get upgrde", reboot --> sound did not work any more

Remark: Done from a fresh image install.around 2021-03-01.

Pearl-852, 2021-03-05:
Since the new release of Raspberry Pi OS from 4th of December 2020, which switches to use the PulseAudio sound server. In some cases, this might affect the audio output settings after the system upgrade.

To correct the problem, see the instructions [HERE](https://www.raspberrypi.org/blog/new-raspberry-pi-os-release-december-2020/) under the sections of _**How do I get it?**_ and _**Updates**_.

Or follow the steps below:
1. Update your system to the latest version
`sudo apt update && sudo apt full-upgrade` <br> (It is safe to just accept the default answer to any questions you are asked during the update procedure.)
2. Install the PulseAudio Bluetooth support
`sudo apt purge bluealsa`
`sudo apt install pulseaudio-module-bluetooth`
`reboot`
3. Swap over the volume and input selector on the taskbar from ALSA to PulseAudio, after your Raspberry Pi has restarted.
a) Right-click a blank area on the taskbar and choose **Add / Remove Panel Items**.
b) Find the plugin labeled **Volume Control (ALSA/BT)** in the list, select it and click **Remove**.
c) Click the Add button, find the plugin labeled **Volume Control (PulseAudio)** and click **Add**.
d) Use the Up or Down button to move the new Volume Control to your desired position.
e) Click the Close button, to close the **Panel Preferences** window.
4. Right-click on the **Volume Control** icon on the taskbar, select **Audio Outputs** and choose **HDMI** if it wasn't ticked.
5. To prevent some applications are ignoring the effect of the PulseAudio output switcher, which probably caused by an old ALSA configuration file still being on the system. Remove it with the following command in a terminal. <br> `rm ~/.asoundrc`

Pearl-852, 2021-03-05:
**Note**:
I have flashed a new [CrowPi2 image](https://drive.google.com/file/d/1kL1lQcXDQit4ITlbM5aiwib_8aIJ9ziC/view) and confirmed with your symptom. After applied the fixes as described above, audio sound works correctly through the internal speakers and the headphone jack on the side .

Dirk1966, 2021-03-06:
We followed the instructions above and it works perfectly again - thanks for the nice technical background links too!

closed 2021-03-06 by Dirk1966