Elecrow-RD/CrowPi2 #46: How to install Crowpi2 software and drivers manually from standard Raspberry Pi OS?

Pearl-852, 2021-07-03:
Instead of using the pre-built CrowPi2 image, I would prefer to install everything (i.e. Crowpi2 software, drivers, etc.) myself. Especially, I have my CrowPi2 system customized with various settings, apps, etc. So, I prefer not to use the pre-built CrowPi2 image when it's updated.

Please provide an installation instructions, similar to what you have provided for [CrowPi1](https://github.com/Elecrow-RD/CrowPi#drivers-installation).

CaptCOBOL, 2021-07-04:
Absolutely need this! Thanks Pearl!

rabagliati, 2021-11-06:
I think I have found out how to do this. I needed the LED seven segment display, but the updated driver I found for it displayed wrong on the third digit. I think this is because Elecrow chose to wire up all the decimals and colons, so their pinout for the LED driver is slightly different.

But, good news, the Crowpi2 install image comes with modern python-3.7 libraries, so I did the following :-

You will find the libraries below in  /usr/local/lib/python3.7/dist-packages/ on the Crowpi2 image. Either
scp these off the image, or use losetup to mount the image on a loopback mount under linux. The LED_Backpack comes out of the Happy Birthday directory. I don't have an image handy, but it is under /usr/local/project/ I think.

- Create a directory ~/.local/crowpi2. We will use this to put the crowpi2 python libraries.
- Into this directory, copy 
- Adafruit_BBIO  Adafruit/         <-- CharLCD needs this, copy the whole directory
- CharLCD-1.1.1-py3.7.egg      <-- a single file
- Adafruit_LED_Backpack/       <-- the whole directory, from the Happy Birthday app

I didn't need to copy the GPIO libraries, maybe because because I had previously installed [adafruit-blinka](http://learn.adafruit.com/adafruit-io-basics-digital-output/python-setup). I *think* the Blinka install allows them to use the C code they wrote for the Arduino without a rewrite. But it might just work if you copy the Adafruit_PureIO-1.1.9-py3.7.egg and add it to PYTHONPATH.

After copying those libraries, add this to ~/.bashrc

`export PYTHONPATH=$HOME/.local/crowpi2:$HOME/.local/crowpi2/Adafruit_CharLCD-1.1.1-py3.7.egg`

log out, log in.

```
python3
from Adafruit_CharLCD import Adafruit_CharLCDBackpack as LCD
crowpi2 = LCD(address=0x21)
crowpi2.set_backlight(0)

from Adafruit_LED_Backpack import SevenSegment
segment = SevenSegment.SevenSegment(address=0x70)
segment.begin()
segment.clear()
segment.print_hex(0x8888)
segment.write_display()

```

Serpentstar, 2022-03-15:
Did this issue ever receive any traction?  I just picked up one of the CrowPi2 units and went straight to downloading the new OS.  I would much rather see official packages made available than to see a proprietary OS.  That is one of the many selling points behind the Pi in the first place.

Thanks.

DmitryBurstein, 2022-05-22:
I second this. Especially now when Bullseye is available for quite some time, but CrowPi2 image for it is still absent.