Elecrow-RD/CrowPi2 #47: No sound from speakers

SpareSimian, 2021-11-22:
Brand new CrowPi2, no hardware changes and no software installed. I got wifi going intermittently and tried to play something on Youtube and hear nothing. Is this likely a software issue with whatever OS version came on the microSD? What hardware connections can I check? Is there a good sound check utility to verify correct operation?

rabagliati, 2021-11-22:
The Crowpi2 is a platform for the Raspberry Pi more than a laptop.

It has no builtin audio. There is a squeaker and a vibrator, but no speaker.
If you want sound, I would recommend a bluetooth dongle and bluetooth speaker.
The "Happy Birthday" app can play  tune on the squeaker, but I would not call that audio.

SpareSimian, 2021-11-22:
I saw the headphone jack and figured there was a DAC involved, and possibly speakers under the front. Or are those heat vents?

Pearl-852, 2021-11-22:
Yes, there are a pair of stereo speakers near the front on the bottom and they do works.

FYI, there is an [official forum](https://forum.elecrow.com/) which monitored by the Elecrow tech supports. So, I reccomend you registering an account and then post your questions/issues there. I don't see them response here often.

Pearl-852, 2021-11-22:
If you are not getting sound out of the speakers, right-click on the **Volume** icon at the top of the screen and check that the **Audio Outputs** is set to '**HDMI**'.

SpareSimian, 2021-11-22:
Thanks. I discovered that the included speaker-test utility works well to test the speakers. My problem ended up being poor wifi (possibly a Pi radio issue) that was cutting out and dropping my Youtube test. 

https://linux.die.net/man/1/speaker-test

For example, try "speaker-test -t sine -c 2 -s 2 -f 1000" to play a 1 kHz tone out the right speaker.

closed 2021-11-22 by SpareSimian