Elecrow-RD/CrowPi2 #52: Crow pi 2 battery type with pi4B 8G that fits "drawer"?

katumbita, 2024-11-05:
I have bought a crow pi 2 deluxe from aliexpress and battery is not included.Read i need this
[image]
Do you guys think this amazon uk link will be good?I am mother technology level jurasic to the point i don't get if this product will have ports required,output and input or what wires to buy so they don't restrict/limit energy to raspberry 4B 8G and screen.
Help please😅
https://amzn.eu/d/2wHCs2Uhttps://amzn.eu/d/2wHCs2U

Pearl-852, 2024-11-06:
I would suggest you to visit the Elecrow official forum, and read the [CrowPi2 FAQ](https://forum.elecrow.com/discussion/87/crowpi2-faq).

There is a **HowTo** relates to your concern listed under the **Power Problem** section, in item 5.

Pearl-852, 2024-11-15:
@katumbita 
This power bank might fit what you are looking for.

[LISEN 0.3" Ultra Slim MagSafe Power Bank](https://www.amazon.com/dp/B0DD74T3LF)